derhopp/Marlin-with-Anycubic-i3-Mega-TFT #12: TFT on RAMPS

BETLOG, 2018-04-07:
Question:
 How easy is it to use your code to drive the anycubic TFT on regular Marlin?
Presumably there is a lot of code in this branch that deals with the weird ways anycubic had the trigorilla wired, but assuming i want to run regular Marlin on regular RAMPS, which parts of your code would be required to enable the TFT? Is this a simple thing to do?

derhopp, 2018-04-07:
Using a RAMPS instead of the Trigorilla-Board should be easy.  AFAIK the expansion header is pin compatible.

The TFT communication mostly serial communication.  Additional features (SD Card and filament runout) are clearly addressed through the pin_XXX.h configuration. Try to compare the pin defines for differences to RAMPS, add missing pin defines.  It might even work with the Trigorilla pin config (it is originally derived from RAMPS).

BETLOG, 2018-04-08:
I assume you have made changes to marlin that are specific to making it work on the trigorilla. Changes that might interfere with or be contrary to what RAMPS would need.
Is that correct?
If so: If i wanted to separate *just* the required files to run TFT on RAMPS... which files would that be?