derhopp/Marlin-with-Anycubic-i3-Mega-TFT #14: Any plans to update to newest bugfix branch?

Murloc992, 2018-07-07:
Hello, as the issue title states are there any plans to update to the newest 1.1.x bugfix branch of marlin?

derhopp, 2018-07-07:
Yes I am planning to do so… but it is quite bunch of work doing that.

Murloc992, 2018-07-07:
I am trying to setup the newest branch without any support for the TFT and so far everything seems to work. Except homing makes the printer call kill(). Weird. Any way to find out what triggered kill()?

derhopp, 2018-07-07:
Hi,

thats unfortunately trial and error.  Did you change your drivers?  Have overstepped (too high step frequency while moving)?  Did you accidentally put a signal to the kill-pin?

Best regards.

Murloc992, 2018-07-07:
I have TMC2208 in 4 out of 5 driver slots. So far I haven't changed much from defaults. My drivers are not connected by UART. Right now I am just trying to replace all kill() instances with readable output..

Murloc992, 2018-07-07:
Okay, I found it. It's endstop validation. After I removed the kill from it I get:
`echo:endstops hit:  X:-5.00 Y:-3.39 Z:-1.99`

I wonder why would it fail?

Murloc992, 2018-07-07:
It's the X axis somehow.  I move it to the center, press home and it "homes" in the middle of the bed. Consequent home attempts makes it "wobble" back to the endstop. Weiiiiird.

M119 shows the endstop as open when it homes at a random spot.

When it homes to the endstop it shows up as TRIGGERED.

derhopp, 2018-07-08:
What does the end stop status show, when the endstop is physically triggered and when not? Could be inverted in the FW, or is it maybe short?

Murloc992, 2018-07-08:
Endstop is perfectly normal according to M119. Steps per unit are the same as default:
`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 92.6 }`

Murloc992, 2018-07-08:
I think it is related to newest firmware and only X axis. I have opened an issue on Marlin repo itself:
https://github.com/MarlinFirmware/Marlin/issues/11224

Whenever I flash back to the firmware on this repo from the bugfix branch everything works just fine.

derhopp, 2018-07-08:
> MarlinFirmware#11224

Okay, that's why I don’t have that behavior, because I am still on the version of my marlin fork.

Murloc992, 2018-07-08:
Seems like making the newest firmware work with Mega will require hardware mods to the endstops themselves. Namely, adding 100nF capacitors in parallel.

derhopp, 2018-07-09:
I don’t put any capacitor nowhere on my trigorilla board.  I just burned my part fan transistor through with a 22uC tantal capacitor.  Fortunately it the transistor is short not open... so the fan still works. (:

Murloc992, 2018-12-09:
@derhopp Thanks for doing the update. :)

closed 2018-12-09 by Murloc992