derhopp/Marlin-with-Anycubic-i3-Mega-TFT #19: print menu frozen in TFT

lander1978, 2018-09-30:
I download the last https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT firmware, the 1.1.x branch, and compile with arduino 1.8.7, mega board option. 
All seems ok, but wend I touch a "print" option in the printer the screen frozen. Nothing that I touch works, only power button. Control with octoprint its ok.

I can print with octoprint, but with this issue, I can't print never with printer alone.

Can you help me? Thanks

derhopp, 2018-10-01:
Please connect the printer with usb (e.g. to octoprint) and send me the log output when you try to print directly (make sure, that you enable ANYCUBIC_TFT_DEBUG at the end of Configuration.h).

I will try to help. (:

lander1978, 2018-10-01:
Hello derhopp, I receive this:

Recv:  T:25.12 /0.00 (975.75) B:26.14 /0.00 (974.50) @:0 B@:0
Recv: TFT Serial Command: A8 S0
Recv: 0<Special_Menu>
Recv: 1/�megaD�
Recv: 2xyzCalibration_cube.gcode
Recv: 3V29D_Fixed.gcode
Recv:  T:25.19 /0.00 (975.63) B:26.14 /0.00 (974.50) @:0 B@:0

thanks,

closed 2018-10-01 by lander1978