derhopp/Marlin-with-Anycubic-i3-Mega-TFT #20: Extra LCD 12864

otherguy, 2018-10-07:
I'm downgrading my 04 Anycubic i3 with an 8bit Trigorilla board to get BLTouch support and better drivers and I'm planning to use your firmware.

Do you know if it's possible to add an extra LCD 12864 display in addition to the i3 TFT display? I'm building an enclosure and would like to mount the 12864 so it's visible from the outside. The headers are there on the Trigorilla board so I assume it's just a matter of enabling it in the firmware?

Any help or info appreciated!

derhopp, 2018-10-07:
There is hardly anything that hasn't been done yet. (:  
Look here... https://www.thingiverse.com/groups/anycubic-i3-mega/forums/general/topic:32269

But... yes there is always a but... I don't know if the TFT from the 32bit one is usable without the board.  If the TFT are the same, then you just need UART3 + Power (either 12V or 5V... I don't know... maybe both) + a DI for the "kill" signal.

otherguy, 2019-01-25:
@derhopp so, to finally resolve this topic ;)

Maybe someone finds it useful in the future.

- The 32bit TFT is different from the 8bit. The connection of the 32 bit board to the TFT is a 0.5 Pitch 40 pin ribbon cable that goes directly to the mainboard. 
- The 32bit mainboard is one board -- no modules. SD card slot is integrated into the mainboard.
- Anycubic sold me the 8bit TFT, the SD card module and the "hub" module including all cables for 29$ shipped. That was worth it even though I don't intend to use the hub module or the SD card module.
- All plugs and connectors are completely different between the 8bit and the 32bit board and need to be changed. For example, on the 32bit board, there is a 8pin single row JST connector from the little board at the side of the base that connects to the endstops and motors. On the 8bit board, there is a 6x3 pin dupont header that does that but it's normally plugged into the "hub" board that simplifies that and connects to that board on the side of the base.

So what I ended up doing: 
- I got one of these [cheap 12864 modules off Aliexpress](https://www.aliexpress.com/item/LCD-12864-Controller-3D-Printer-ramps-1-4-Controller-LCD12864-Display-Monitor-Motherboard-Blue-Screen-Module/32961365080.html) for like 8$
- The adapter board that comes with that module is not needed
- I plugged the 12864 LCD into the `EXP1` and `EXP2` ports, installed the [`U8glib`](http://code.google.com/p/u8glib/wiki/u8glib) into my Arduino libraries folder and enabled the LCD in Marlin's `Configuration.h`: `#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
- Now with the 12864 plugged in, the "hub" board does not fit. So I removed the hub board and the SD Card board (the SD card slot is now on the 12864 anyway).
- To plug in the TFT display, I removed the 5 pin JST connector (on the hub side) and reworked the wire into 3 dupont connector:
  - 1 2-pin dupont connector for the green (`5V+`) and yellow (`GND`) wire going to a free 5V / GND pin on the board.
  - 1 2-pin dupont connector for the grey (`RX`) and blue (`TX`) wire going to pin 15 and 14 respectively
  - 1 1-pin dupont connector for the red wire that _should_ go to pin 41 (`KILL_PIN`). This is used for the _reset_ feature of the TFT. I'm not really using that feature and the 12864 has a reset / kill button as well so I have not connected the cable yet. I'm considering running it to my Raspberry running Octoprint and from there doing a full reset of the whole system.
- Now both the 12864 and the TFT screen are working but since the hub board is missing, the connection to the hub on the side of the base is not plugged in. Since I was coming from a 32 bit board anyway, and my cable did not fit, I just reworked the 8pin single row JST connector (that's the 6pin JST one going to the hub board on the 8bit board) and the 3pin JST for the Y endstop into 6 3-pin dupont connectors and plugged them into the 6x3 header for the endstops/limit switches.

Voila! Conversion from 32bit to 8bit done and both TFT and 12864 LCD working! 🎉😁 

Thank you @derhopp for your amazing work getting the TFT running!

closed 2019-01-25 by otherguy

derhopp, 2019-01-26:
Sounds like a cool project.

I won't be able to use a 12864 LCD because my GPIOs are eaten up by my second extruder and my TMC2130s via SPI.

otherguy, 2019-01-26:
Yeah it was more of a proof of concept. I still need to wire my TMC2208 via UART and I guess I already don’t have enough pins left for that.

simo26246, 2019-05-22:
> 
> 
> Yeah it was more of a proof of concept. I still need to wire my TMC2208 via UART and I guess I already don’t have enough pins left for that.

Hey @otherguy I'm in the process of downgrading from the 32 bit to the 8bit for similar reasons, don't recon you'd be able to send me a pin diagram for the way you've wired it? I'm a visual person, haha.

otherguy, 2019-05-22:
Don't know how much help that's going to be because you have to remap everything in Marlin anyway. I also skipped the 

That's how it looks like now. As you can see, I skipped the hub board and pulled apart the LCD connector:
[image]

Basically all you need is this is a picture with pin mapping, available at:
http://www.xcsource-pic.com/TE627-E-10-9.jpg

A picture with port descriptions is at:
https://www.thingiverse.com/download:4012609

Meganium97, 2021-11-06:
> @derhopp so, to finally resolve this topic ;)
>
> Maybe someone finds it useful in the future.
>
> * The 32bit TFT is different from the 8bit. The connection of the 32 bit board to the TFT is a 0.5 Pitch [...]

Sorry if I’m hopping on a bit late but I assume that this works if I wanted to just do without the tft screen? Would I just skip the last two steps?

Gregbury, 2022-05-18:
Bonjour
J ai installé un tft35 mz v3 .
Pour remplacé LCD d origine le tactile fonctionne en me branchant TX et rx au 2 résistance la fiche hub j'ai brancher en direct sur les fiche de la carte pour avoir la place pour brancher les fiches exp1 et 2 ,mais je n' arrive pas a compiler en décrochant les 2// de la ligne reprap_discount... Sur une mega x

Gregbury, 2022-05-18:
Bonjour
Je possède une mega x j'ai installé un tft mz v3 le tactile je les connecté au rx0 et tx0 ok ça marche mais je n' arrive pas a compiler pour activer les exp1 et 2

nickademus420, 2024-01-30:
Hi. So I have done all the wiring mods to make it so I can use the original screen and the mini 12864, but I need to upload the new Marlin firmware. I have the Marlin.INO loaded, selected the Arduino Mega or Mega 2560, and the processor as ATMEGA2560. I selected the right com port, and tried to upload, but keep timing out. Any tips?