derhopp/Marlin-with-Anycubic-i3-Mega-TFT #21: Pre-heat causes Z raise

ollie-campbell, 2018-10-08:
### Description

Using the Pre-heat options via the LCD causes the Z axis to raise.

### Steps to Reproduce

1. Tools
2. Pre-heat
3. Pre-heat PLA

**Expected behavior:** [What you expect to happen]
Just the hotbed and nozzle to heat to the set temperatures.

**Actual behavior:** [What actually happens]
The hotbed and nozzle do pre-heat, but the Z axis raises.
[Configuration.zip](https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT/files/2455148/Configuration.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

newPOPE, 2018-10-08:
Is it only for PLA preheat?

I think that is normal for Z raise. One of the reason can be that hot filament can drop down from nozzle. Why is the issue for you?

ollie-campbell, 2018-10-08:
Hello,
I haven't tried the Pre-heat for ABS but will try when my print has finished.

It is not an issue, it is just different from the standard firmware that ships with the printer and presumed that it must be a bug.

newPOPE, 2018-10-08:
:) for me is a good access from printer => no issue. Maybe @derhopp will know more.

derhopp, 2018-10-08:
This feature is meant to protect your bed surface. If z<10 and you enter preheat on the TFT the axis rises.  

Working as intended —> no bug. Closed.

closed 2018-10-08 by derhopp