derhopp/Marlin-with-Anycubic-i3-Mega-TFT #24: m600

ondrax1, 2018-11-20:
Hello, i have question. I have Anycubic with derhopp marlin 1.1.x I update original FW, becouse i need function m600 and changing filament. But i have problem with this. I setup
#define ADVANCED_PAUSE_FEATURE

and

#define NOZZLE_PARK_FEATURE

After check, arduino ide wrote me #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller or EMERGENCY_PARSER." And i setup function Emergency parser. After that i flash this FW to my printer. Pausing is working! (fantastic) but i dont know how i can continue print, after change filament. If i try make pause, temperatures goes to 0 and pring progress is stopped.

video here: https://youtu.be/RSgZyg1jOtM

(link here https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT)

Sorry for my bad english.

Thank you so much

newPOPE, 2018-11-20:
I think that display fw doesn't support m600 feature although printer fw support that.
Did you try some workaround? Maybe somethink like this:

- use m600
- when nozzle goes to pause mode
- change filament
- click pause on display
- click continue (display support)

but this is only idea, maybe it will not work.

derhopp, 2018-11-20:
Actually it does work but not through the display.

You have to send a M108 through the terminal to continue.

In case you connected your printer with OctoPrint, you have to send (one or two) FakeAcknowledgements in between.

closed 2018-11-20 by derhopp

ondrax1, 2018-11-21:
Hello, i use pronter face. M600 in gcode make me pause, i change filament. Send m108 via ptonter face and problem 
[image]

ondrax1, 2018-11-21:
enable "NOZZLE_PARK_FEATURE" (Configuration.h), "ADVANCED_PAUSE_FEATURE" (Configuration_adv.h) and "EMERGENCY_PARSER" (Configuration_adv.h).