derhopp/Marlin-with-Anycubic-i3-Mega-TFT #26: Setting up TCM2208 Drivers

CaptInsano, 2018-12-06:
As per most of the advice out there on the Anycubic i3 Mega, I have changed out the stock stepper drivers for the TCM2208. They are working perfectly under the stock firmware and are much quieter that the stock drivers.

I am now looking to change from stock firmware to Marlin to take full advantage of these drivers.

I have installed the TCM2208 library and under configuration.h,  I have changed the driver settings:

`
#define X_DRIVER_TYPE  TMC2208

#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
#define X2_DRIVER_TYPE TMC2208
#define Y2_DRIVER_TYPE TMC2208
#define Z2_DRIVER_TYPE TMC2208
#define E0_DRIVER_TYPE TMC2208
`

When compiling however i get the following errors:

`stepper_indirection.cpp:299:32: error: 'X_SERIAL_RX_PIN' was not declared in this scope`

followed for similar errors for Y, Z, Z2, E0 etc.

I have looked and looked but cannot find any obvious answer to this.

Where do I define the correct pins?

newPOPE, 2018-12-06:
What is the reason for change firmware? I just change drivers, turn motors pins and setup voltage to 1.2V and everything works. As firmware I am using my fork of this because need to correct XYskew factor.

derhopp, 2018-12-06:
You have to edit pins_TRIGORILLA_14.h .

From my experience it is worth utilizing the features of the communication to the stepper driver. E.g. hybrid mode or diagnostics... and much more.

closed 2018-12-06 by derhopp

derhopp, 2018-12-06:
Simply add the necessary defines of the pin you have used for connecting the UART to the trigorilla to pins_TRIGORILLA_14.h yourself. But make sure that these pins aren’t used, as some processor pins have multiple connectors on the board or/and are not listed in the pins file.

The default firmware doesn’t have the defines, as the stock printer doesn’t have TMCs.

otherguy, 2018-12-06:
@derhopp did you do that on yours? Any chance you could show a photo of where you connected the UART pins?

derhopp, 2018-12-07:
I don't use TMC2208s but I use six TMC2130 with full SPI access.  My I use the following pins... https://drive.google.com/open?id=1Fy8QxFAcTLMPc_V28V5CENHZI3NIAREf