derhopp/Marlin-with-Anycubic-i3-Mega-TFT #28: Allgemeine Frage

TigerClaw33, 2019-02-05:
Hi, 
ich hätte eine allgemeine Frage. Ich würde gerne die offizielle Firmware von Anycubic nutzen, um mein 8 Bit Board vom Anycubic I3 Mega etwas umzuprogrammieren. Leider gibt es dort nur die HEX-Datei. Wo kann ich die einzelnen header und source files herbekommen? 

Ich vermute mal, dass die nicht verfügbar sind und du deine Version aus der Marlin-Version generiert hast und vermutlich auch etwas Zeit investiert hast, um das TFT Display zum laufen zu bekommen. Ist das richtig? Musstest du bei der Anpassung der Marlin Firmware auch alle Werte für die Schrittmotoren manuell anpassen? Warum teilt Anycubic die files nicht?

Im Prinzip könnte man jede Marlin-Fassung direkt flaschen, müsste eben nur die Werte für die Motoren und die Werte der Temperaturfühler anpassen. Ist das richtig? Ich bin da recht neu in dem Thema.

peace4all, 2019-03-08:
geteiltes Leid... aber hier: https://github.com/ANYCUBIC-3D/I3-MEGA das sollten die Originale sein, basierend darauf gehts sicher ohne aufwändig unvollständiger Doku nachzurennen... Ja manche können das einfach super, aber bringens nicht hin, in den Sourcen alles vollständig zu kommentieren... hab' damit leider auch schon Tage wertvoller lifetime vernichtet.
Wennst die Stepper austauscht, ist es mgl. schon sinnvoll, den Treibertyp zu ändern..
und ja, bei #define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 92.6 }
machtes auch Sinn die Werte zu verdoppeln bzw. den Materialvorschub überhaupt neu zu vermessen -
Ich weiss zwar jetzt nicht ob es bei den TMC2208 nicht noch mehr Schritte sind als bei den von mir eingesetzten DRV8825 ern... kann aber auch sein dass in der configuration.h.adv die Anpassung schon vorgenommen ist, das hat sich aber bei den bisherigen Compilaten nicht bestätigt, der Drucker nutzte da immer nur ein viertel des Bed's ...  :1234: 
BR cu

AndiP1234, 2020-05-05:
https://github.com/davidramiro/Marlin-Ai3M/releases/download/v1.4.6/Marlin-Ai3M-v1.4.6-TMC2208.hex

https://github.com/davidramiro/Marlin-Ai3M

After obtaining the hex file:
Flash the hex with Cura, OctoPrint or similar
Use a tool with a terminal (OctoPrint, Pronterface, Repetier Host, ...) to send commands to your printer.
Important Connect to the printer and send the following commands:
M502 - load hard coded default values
M500 - save them to EEPROM

If you are using this on a Mega-S, those four additional commands are necessary:
M92 E384 - set correct steps for the new extruder
M203 E30 - limit extruder feedrate
M204 R1500.00 - lower retract acceleration
M500 - save the values
I highly recommend calibrating the extruder.

Ich musste noch ein Autotune vom Hotend durchführen. Bei mir hat es mit 75% des Bauteillüfters nicht ganz funktioniert, machs lieber mit 50%. Danach beobachte der Hotend Temperatur beim Drucken.