derhopp/Marlin-with-Anycubic-i3-Mega-TFT #30: Remove Special Menu to save memory

gremorox, 2019-04-19:
Hi, great work. But I have a question. I use a Reprap LCD as second screen. Is it possible to remove the special menu to save memory.?

Wanto use

derhopp, 2019-04-19:
Unfortunately the special menu has no CONFIG option.

gremorox, 2019-04-20:
OK. Thank you for the fast answer

gremorox, 2019-04-21:
Is there an overview of the changes you made? Because i want to merge with the actual Marlin 2.0 bugfix files. 

Is it only : AnycubicSerial.h et cpp / AnycubicTFT.h et cpp / and the changes in Configuration.h, Marlin.h, MArlin_main.cpp?

Thank you for your help

derhopp, 2019-04-23:
And Configuration_adv.h  ... then it is it.

shoex, 2019-04-23:
Thanks a lot

kikagithub, 2020-03-02:
Hi, did you get Marlin 2.0_bug fix working with Anycubic I3 mega?
I tried it but I always got compiler errors.
Could you please share source files?
Thanks

shoex, 2020-03-02:
> Hi, did you get Marlin 2.0_bug fix working with Anycubic I3 mega?
> I tried it but I always got compiler errors.
> Could you please share source files?
> Thanks

Hi, 
I got Marlin 2.0 working, but I decided to remove the TFT completly from the printer. So I compiled directlyy from the marlin source without the anycubic part.

The source here worked also for me, but it takes too much memory with reprap LCD.

What is your compiler issue ? You use also an Reprap LCD? Or both ?

kikagithub, 2020-03-02:
Hi,
thank you for your prompt reply.
When I compile I get tons of redefines and some missing files...
Now I'm using the Marlin-Ai3M-2.0.x-2.0.x with the original TFT display
version modified to use BLtouch, Nozzle Clean, TMC2208_standalone and SD.
The reason I'm trying to upgrade to the 2.0.x_bug_fix is that I was trying
to upgrade my current Marlin version with  TMC2208  drivers to tmc 2209
(with serial features) but, even if everything works fine, they motor
steppers became very noisy and I've found probably there is a bug in the FW
that should be fixed with the 2.0bug_fix version or 2.1.
That why I'm trying to upgrade to 2.0.x bug_fix or better 2.1.
I copied the dedicated source files (serial and TFT) and  I spent more than
5 hours with winmerge to find differences in the configuration files (the
new ones has a lot of changes), but at the end of the day I went back to
the TMC2208 and Marlin-Ai3M-2.0.x-2.0.x waiting for a more suitable release
version or more time to play with!
Thank you for your help

shoex, 2020-03-02:
I think it is hard to merge this with the original marlin Bugfix branch, because u would not have enough memory to do every thing u want.

But if you provide ur files, and the exact source, I will have a look for the errors, when i got time.

kikagithub, 2020-03-03:
Hi, yes it is very hard in particular the porting of the TFT code.
I'll keep you updated as soon as I've got a slot of time to work on it
Thank you for your support and willingness to help me